qian256/HoloLensARToolKit #2: Matrix code does not match original ARToolKit

qian256, 2017-02-27:


Quach90, 2017-02-27:
Is there a way to decode the lib files, to see which ID's matches which matrices? Or where would this problem be fixed?

NattyBumppo, 2017-03-13:
I've also observed this issue myself. How can we know which IDs match which matrices?

qian256, 2017-03-13:
@NattyBumppo, I figured out a work-around temporarily, but I will fix the problem hopefully in two weeks.
So, currently, for matrix code 3x3, marker with odd ID (x) is changed to (x+31); marker with even number ID is kept same. If you look at the configuration file I made for the cube (cube.dat), you will observe the difference.
This is definitely a bug somewhere in the ARToolKitUWP dll I build. I will look into that in detail when I got time.

NattyBumppo, 2017-03-13:
Gotcha. I had already noticed that even numbers stay the same :) I had seen your cube.dat and was wondering why you assigned the IDs the way that you did. Now I understand, thank you!

qian256, 2017-05-14:
@Quach90 @NattyBumppo I just pushed up version 0.2 of this project, where the mismatch between matrix ID is fixed.
The reason for the previous issue is that: the image RGBA data for tracking was provided by Unity WebcamTexture object, which uses the Unity's convention of pixel coordinates (x axis pointing right, y axis is pointing upward). But in ARToolKit, OpenCV, or OpenGL, the pixel coordinates are: x axis pointing right, while y axis pointer down.
Therefore, when I feed the RGBA array obtained from Unity to ARToolKit, the image is "flipped upside-down".
Then, if we look at the 3x3 matrix marker, the markers with even ID number are actually horizontally symmetrical, so these markers ID do not change. The markers with odd ID number are not horizontally symmetrical, but their mirrored ones also exist in the marker list! The mirror of marker 1 is 32, and the mirror of marker 3 is 34, etc.
Now it is fixed because in v0.2, the image data does not come from Unity WebcamTexture anymore, but from Windows.Media.Capture instead. The pixel coordinate system is consistent between Windows Media APIs and ARToolKit.
Thanks!

closed 2017-05-16 by qian256