qian256/HoloLensARToolKit #3: Support of NFT (Natural Feature Tracking)

qian256, 2017-02-27:


janholo, 2018-04-03:
Hey
I need this feature and want to implement it. Could you hint me in the direction how this can be done?

qian256, 2018-04-03:
Hi @janholo , thanks for your interest! The NFT needs to involve some other dependencies. I think the direction is similar with the current setup: 
1. Build original ARToolKit libraries for UWP, excluding the video IO part. 
2. Create a wrapper or modify on current wrapper: ARToolKitUWP folder in my repo, to expose function calls for creating NFT marker, and expose the option to specify NFT tracking instead of barcode or pattern tracking. 
3. Modify the unity package so that user can specify options for NFT
4. A sample scene to demo it.

actarus1975, 2021-06-12:
Hi @qian256, first of all thanks a ton for the great job on porting ARToolkit 5 to HoloLens 1 & 2!

Finally, following your precious tips, I've been able to add the NFT support to your project!
Last week I tried to reach you at your public email to inform you, but with no luck.
I forked this project and created a new branch named "nft_detection". Next monday I'll create a pull request and provide you with all the required details.
Thanks again!

qian256, 2021-06-13:
@actarus1975 wow that's fantastic! Looking forward for the pull request!

actarus1975, 2021-06-16:
Hey @qian256, the pull request is ready if you need it!