qian256/HoloLensARToolKit #5: "Magic function"

jdcart, 2017-03-03:
I've been reading your code and I can't seem to understand what you are doing with your "magic function". Do you mind sharing or commenting on what is the approach that you are taking on this, and also where are you getting all the values for "magicMatrix1" and "magicMatrix2" from (considering you are hard coding those values in)?

Also, I was reading your blog and you mentioned that all of your code is using left-handed coordinate systems. If I understand correctly you are using "LHMatrixFromRHMatrix" to achieve this, however, in your code this specific function is commented out. Are you ignoring this function and changing the coordinate system in your own way?

As an extra note, I want to point out that when you use arunity all the frames are already changed to work properly inside of Unity and there is no need to modify the marker configuration files for them to be displayed correctly.

qian256, 2017-03-03:
The magic function is the calibration of position and rotation, that takes tracked result to the correct visualization. If you turn it off, you will observe that the overlay is registered with the user. Of course because there is no guarantee that the "tracking space" is same as "display space". There is offset in position, rotation, scale (even skew). The current magic calibration is one that fits me. I obtained these values from a calibration procedure (I will push them up around mid March).
For HoloLens, different people with different IPD might have different calibration matrices. I am trying to create a friendly calibration procedure (another Unity scene) for arbitrary user to find their best calibration matrices via several alignment tasks.
Currently I don't call the conversion function because it is already correct. Your extra note about the coordinates is exactly what I tried to find out. Thanks a lot!

jdcart, 2017-03-03:
Thank you for explaining! everything makes more sense now.

closed 2017-03-03 by qian256

araujokth, 2017-06-20:
I was trying to understand the impact of the magic function on the position of the virtual object and had a question. 

From what I understood, the target object is the one that is being displayed, which is obtained by transforming the dummyGameObject into the world coordinates. The dummyGameObject is then calculated via the position and rotation from latestTransMatrix. I was then trying to understand the impact of the magic functions on the object and just as a simple trial I defined the 4th column of magicMatrix1 as [0.1;0.1;0.1;1.0] and to see what would happen. I then printed the values for the position of both the dummyGameObject and the target before and after applying the magicMatrix1. While the position of dummyGameObject gets translated by [0,1;0.1;0.1], the position of the target gets translated by another value. 

I was then wondering what was happening with the transformation of the target? is it that the translation operation gets modified when transformed into world coordinates or did I miss something? 

If one would like to then calculate the transform required to align both the real and virtual objects as you mentioned above and looking at your paper, both pi and qi would be acquired from the target positions right? But if such calibration transformation gets modified afterwards, then I guess that would destroy the alignment since the final transform is not the one intended? 

Thanks for the help!

qian256, 2017-06-23:
Hi @araujokth , 
The purpose of the dummyGameObject is to act as an **empty object**, that relays the transformation of tracking to the world coordinate system. dummyGameObject is the child of HoloLens Camera, but **target** is in the root. The purpose here is to make use of the SLAM of hololens. When the tracking result stops update, the **target** will be fixed in the world instead of fixed in front of the camera. This is tuned by the parameter **AnchorToWorld**.

araujokth, 2017-06-24:
Just to clarify, what I was meaning was if the magic function should be applied to the target position (after anchortoworld is applied) and not be applied to the latestTransMatrix like it is now, since during a calibration procedure pi and qi is obtained from the target in world coordinates. Thanks again!

araujokth, 2017-06-26:
Not sure why I thought that you could only do this step based on just world coordinates. Of course you can do it from both coordinates, but then you have to transform it correctly. Sorry for the misunderstanding!

qian256, 2017-06-27:
@araujokth , you are right about this point. We assume the display coordinate system is euclidean w.r.t to world coordinate system, but the tracking coordinate system is a little bit off. So the result of tracking needs to be parsed through a calibration function to be suitable for display. Also, the implementation in the repository and the implementation in the paper is not exactly same. Here in the repository, the calibration, or the magic function is separate for position and rotation, it is just a trick that makes it easier to tune.