qian256/HoloLensARToolKit #6: Front camera cannot get work

mudusun, 2017-03-07:
when I rebuild the scene in my own project, why the camera cannot get work? I made every setting as same as it was done in this project. There is no error in deploy but the camera indicator was not on when running the solution. There are no other access package except HoloToolkit and ARToolKitUWP.unitypackage. Can you help me with this? 

[image]

qian256, 2017-03-12:
Not sure what the problem is. Have you solved it?

mudusun, 2017-03-13:
Not yet. Still working on it. And I have no clue...

jdcart, 2017-03-15:
Did you make sure you have the WebCam capability on? 

Player Settings > Windows Store > Publishing Settings > Capabilities

mudusun, 2017-03-16:
Yes,  the WebCam capability is on. I guess that there should have some connection with ARToolKitUWP folder in the project?

qian256, 2017-03-16:
No, the ARToolKitUWP folder builds the dynamic link library `ARToolKitUWP.dll`, and it is already in the Unity project under `Plugins/`

closed 2017-03-25 by qian256