qian256/HoloLensARToolKit #7: A problem with files .dat?

NawelNima, 2017-03-13:
Hello, I tried to change the .dat files of the project with mine, but it does not seem to work. The .meta files are not recreating. 
Can someone help me please?

qian256, 2017-03-13:
Which dat file? camera calibration or multi-marker configuration?

NawelNima, 2017-03-13:
About multi-marker configuration, when I changed them, nothing happens. As if they were not taken into account.

qian256, 2017-03-13:
Are you modifying one of the example? Could you please send me the configuration file you changed, and the screenshot of you ARUWPController and ARUWPMarker options in Unity?

qian256, 2017-05-14:
Hi @NawelNima , did you fix the problem yet?

NawelNima, 2017-05-14:
Hello, not yet. 
I tried to change the data (the postions) in cube.dat or multi-barcode-4x3.dat. But like I said before, nothing changes, the markers in the scene remain at the same place.

qian256, 2017-05-14:
The meta file is still no creating? Have you tried the recent commit, where the marker configuration is same as ARToolKit.

tsteffelbauer, 2017-05-24:
Try to reload the marker by temporarily changing to a different marker and then switching back

closed 2019-01-11 by qian256