qian256/HoloLensARToolKit #9: Not getting captured image taken by camera

seonjoo1031, 2017-03-27:
Hi!
First, thank your for open sourcing your awesome code.
I'm absolutely new beginner to Unity, ARToolKit and Hololens.
As for my situation, I have to make hololens detect some object in augmented reality(this object will be the same as marker) and make something appear around or on this marker.
So I thought I have to use opencv at first, but after searching I found your code and decided to use it.

But this is the first time, I had a problem even at the basic step..
The following image is taken on hololens emulator.
[image]

I walked through the process for running the sample.
I think I followed all the steps you mentioned.
But I just got the result above.
I think the camera is not working correctly because I got a white box in which there was captured image taken by camera in the sample video.. (my laptop has camera)
Is there any other configuration I should have set?

I think I miss something so basic but I'm truly new to this, so I want you to help me...
I'm waiting for your answer.
Thank you for reading.

mudusun, 2017-03-27:
I also have this problem. I don't know why. I believe that  there is something we don't know and we missed but I don't know.

mudusun, 2017-03-27:
I found the solution for this problem. when you import the ARToolKitUWP.unitypackage, don't change the three files: csc.rsp, gmcs.rsp and smcs.rsp. If you already import the package, you need to replace them with  correct files. I think there is something wrong with the three files in ARToolKitUWP.unitypackage, but I don't know why. Any way it works. you will see the camera.

qian256, 2017-03-28:
Hi @seonjoo1031 , the current version of HoloLensARToolKit does not support Emulator very well. One of the reasons is that the video configuration is manually set to 896x504, which corresponds to the recommended setting of HoloLens. However, most video camera does not have such configuration.

I suggest you try deploying directly to HoloLens, following the instructions. I should work very well.

@mudusun , the files `csc.rsp, gmcs.rsp, smcs.rsp` have nothing else than the parameter `-unsafe`. These files should be put in the asset root folder. But there should be no difference between the `.rsp` files in the unity package and the files elsewhere.

qian256, 2017-03-28:
Moved to #11

closed 2017-03-28 by qian256

seonjoo1031, 2017-03-28:
Thank for answer. I'll try deploying with HoloLens directly.