qian256/HoloLensARToolKit #12: Controlling camera feed - issues with stopping and restarting WebCamTexture

thezelus, 2017-04-02:
Continuation of discussion from - #8  
(background)
WebCamTexture doesn't stop automatically and hence I have been experimenting with it and have added two methods to (ARController.cs script) in my local clone of the repository

```
 public void stopWebcam()
    {
        if(webcamTexture != null)
        {
            webcamTexture.Stop();
        }
    }

    public void restartWebcam()
    {
        if(webcamTexture == null)
        {
            startWebcam();
        } else
        {
            webcamTexture.Stop();
            webcamTexture.Play();
        }
    }
```
Below are my observations:
- Calling `stopWebcam()` does stop the web camera and this works fine the **first** time.

- When trying to start the camera after stopping it from the code results in no video feed, which I suspect is due to being indirectly related to this issue: [Unity/Linux - Unity does not free webcam when stopped](https://github.com/artoolkit/arunity5/issues/115)

- I also tried to destroy the object that has ARController and ARMarker scripts attached, this does solve the issue but is a hack and definitely not a permanent fix

hixe33, 2017-04-24:
Hi,
I got the same issue. Did you manage to re-start a tracking feed after destroying the GameObject ? 
I tried to create another GO with an ARController+ARMarker attached, but I still couldn't resume marker tracking.

thezelus, 2017-04-24:
Hello @hixe33,

I did try it but no luck as the solution is finicky (there might be a race condition) and fails randomly. I think the problem is with ARToolKit itself and not specifically HoloLensARToolKit. As I was pressed on time I used Vuforia to implement the desired functionality.

hixe33, 2017-04-24:
Ok, that's good to know. I'll keep looking but thanks anyway !

qian256, 2017-05-14:
@thezelus @hixe33 Camera feed control is available with the latest commits (the newer version v0.2).
Version 0.2 involves a lot of re-write of the video pipeline. Now it is using Windows.Media.Capture APIs instead of Unity WebcamTexture to directly control the camera, including frame size, frame rate, initialization, starting and stopping.
You can find Pause() and Resume() function in ARUWPController.cs now.

hixe33, 2017-05-17:
Indeed, no more framerate drop for me that way. Great work !

closed 2017-05-17 by qian256