qian256/HoloLensARToolKit #13: UNetWeaver error

tsteffelbauer, 2017-05-12:
Hi,

when I open the sample project in unity (5.5.1f1) I get the following errors, which i have no idea how to get rid of. I can't find any networking in the scripts. Does someone have a clue? 

error1:
UNetWeaver error: Exception :Mono.Cecil.AssemblyResolutionException: Failed to resolve assembly: 'UnityEngine.Networking, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null'
  at Mono.Cecil.BaseAssemblyResolver.Resolve (Mono.Cecil.AssemblyNameReference name, Mono.Cecil.ReaderParameters parameters) [0x00000] in <filename unknown>:0 
  at Mono.Cecil.BaseAssemblyResolver.Resolve (Mono.Cecil.AssemblyNameReference name) [0x00000] in <filename unknown>:0 
  at Mono.Cecil.DefaultAssemblyResolver.Resolve (Mono.Cecil.AssemblyNameReference name) [0x00000] in <filename unknown>:0 
  at Mono.Cecil.MetadataResolver.Resolve (Mono.Cecil.TypeReference type) [0x00000] in <filename unknown>:0 
  at Mono.Cecil.ModuleDefinition.Resolve (Mono.Cecil.TypeReference type) [0x00000] in <filename unknown>:0 
  at Mono.Cecil.TypeReference.Resolve () [0x00000] in <filename unknown>:0 
  at Unity.UNetWeaver.Weaver.ResolveMethod (Mono.Cecil.TypeReference t, System.String name) [0x00027] in C:\buildslave\unity\build\Extensions\Networking\Weaver\UNetWeaver.cs:1109 
  at Unity.UNetWeaver.Weaver.SetupTargetTypes () [0x00826] in C:\buildslave\unity\build\Extensions\Networking\Weaver\UNetWeaver.cs:1398 
  at Unity.UNetWeaver.Weaver.Weave (System.String assName, IEnumerable`1 dependencies, IAssemblyResolver assemblyResolver, System.String unityEngineDLLPath, System.String unityUNetDLLPath, System.String outputDir) [0x0004b] in C:\buildslave\unity\build\Extensions\Networking\Weaver\UNetWeaver.cs:1713 
  at Unity.UNetWeaver.Weaver.WeaveAssemblies (IEnumerable`1 assemblies, IEnumerable`1 dependencies, IAssemblyResolver assemblyResolver, System.String outputDir, System.String unityEngineDLLPath, System.String unityUNetDLLPath) [0x0006e] in C:\buildslave\unity\build\Extensions\Networking\Weaver\UNetWeaver.cs:1837 
UnityEngine.Debug:LogError(Object)
Unity.UNetWeaver.Log:Error(String) (at C:/buildslave/unity/build/Extensions/Networking/Weaver/Program.cs:20)
Unity.UNetWeaver.Weaver:WeaveAssemblies(IEnumerable`1, IEnumerable`1, IAssemblyResolver, String, String, String) (at C:/buildslave/unity/build/Extensions/Networking/Weaver/UNetWeaver.cs:1844)
Unity.UNetWeaver.Program:Process(String, String, String, String[], String[], IAssemblyResolver, Action`1, Action`1) (at C:/buildslave/unity/build/Extensions/Networking/Weaver/Program.cs:34)
UnityEditor.Scripting.Serialization.Weaver:WeaveUnetFromEditor(String, String, String, String, Boolean)

error2:
Failure generating network code.
UnityEditor.Scripting.Serialization.Weaver:WeaveUnetFromEditor(String, String, String, String, Boolean)

qian256, 2017-05-14:
Hi @tsteffelbauer ,
I think in HoloLensARToolKit, there is no dependency on networking. Did you include other project as well?
Also, did you add the capabilities of networking in the project manifest file?

tsteffelbauer, 2017-05-15:
Hi,

I figured out, that this bug is not related to your project, but a problem caused by my VM setup. It just happened to be your project where it occurred for the first time.
My apologies.

closed 2017-05-15 by tsteffelbauer