qian256/HoloLensARToolKit #18: Hololense calibration

MichaelHoenes, 2017-06-21:
Hello Long Qian,

first, thank you so much for developing this kit. 
Is it necessary to calibrate each hololense I'm using separately or can I just use the calibration parameters you provided with the kit?

Thank you very much.
Michael

qian256, 2017-06-23:
For the camera calibration, my guess is that we do not need to calibrate for each HoloLens.
But what is more important is the calibration between the display and the tracking. The **magic function** in ARUWPMarker.cs does this. For best visualization (virtual object is aligned with marker), we have to manually tune the numbers here.

MichaelHoenes, 2017-06-24:
Thanks for the fast reply. I'll do so.

ericalbers, 2017-08-12:
Perhaps theres a way to automate the calibration....have a box appear and you move your head to make the standard cube fit perfectly inside, then tap and it would autocalibrate and populate the magicmatrix?

qian256, 2017-08-14:
@ericalbers I have done this test before, but the result is not very consistent. Because:
1. human alignment is error-prone. Many applications let human align many times and find the optimal fit.
2. If one cube is manually aligned, it only gives us an isometric calibration (6 DOF). But consider the situation that there is a slight scale difference: for example, 1 meter is actually 0.95m for hololens display. So it actually gets quite complicated.
I have one **unofficial** paper: https://arxiv.org/abs/1703.05834 that describes some issues and solutions, but the source code is not on Github yet.

closed 2019-01-11 by qian256