qian256/HoloLensARToolKit #19: Object tracking

araujokth, 2017-06-27:
Hi,

I have been using the HololensARToolKit a bit more now and noticed that even though the tracking latency is around 30-40 ms and the rendering between 40-50 ms, the total latency between object detection and overlay of the virtual object on top of the real object appears to be around 500 ms to 1 s. I noticed that particularly when moving the head from side to side and noticing that the virtual object lags behind the real one for a quite long time, or when one suddenly moves the object. I was wondering if you had looked into this issue? I guess that there is quite some unaccounted time between acquiring the image and display of the virtual object for that image, besides the rendering and tracking latency you are calculating at the moment? 

Would be interesting to see if such time could be reduced, but perhaps the best would have to implement a filter taking into account the head pose from the Hololens (or the IMU data directly once it gets exposed in the API).

qian256, 2017-06-27:
The end-to-end latency is a problem, and it is quite hard for us to address, especially within the framework of Unity. As you may noticed that the video, tracking and rendering are decoupled, it introduces higher frame rate of rendering to have a good AR experience, but it does include some delay between different loops. For example, the tracking result is calculated but not displayed at once.
There is also latency from motion of object to the capture. You can try to display the video directly, and you may find that issue too.
As far as I know, IMU data is not directly available to developers, but the spatial mapping can be used for filtering, if the object is relatively stable in the room.

araujokth, 2017-06-27:
Right, I was mentioning the filter assuming that the object is static or moving slowly, or that the object would have an IMU itself. Yeah, it would be interesting to understand what is the lowest latency one could get in Unity, understanding where the extra time is being spent and if that is possible to be optimized.

closed 2019-01-11 by qian256