qian256/HoloLensARToolKit #23: Stop execution of HoloLensARToolkit.

Baxab, 2017-08-11:
Hi!
I was wondering what could be the best way to stop the tracking and the execution of HoloLensARToolkit. I am using your code as one part of my application and at some point I need to stop its execution and free the resources, in order to continue with my application (e.g.: I need to access the webcam, but I can not do it if I do not stop HoloLensARToolkit before). Thank you!

qian256, 2017-08-11:
Currently, the ARUWPVideo can pause/resume. Have you tried if the camera can be accessed from your app when the video is paused? What API are you using to access the webcam? If you are using Windows UWP MediaCapture API, you should be able to insert your functionality into the frame receiving callback without turning off the webcam. If you are using Unity WebcamTexture API, I do not know how the resources are shared (My guess is MediaCapture for HoloLensARToolKit needs to be totally stopped).

Baxab, 2017-08-12:
I am using Unity VideoCapture class. Yes, it can access the webcam when HoloLensARToolkit is paused, but the problem is that I can not know when the resource will be available, since the Pause() function is async and it takes a while for it to finish its execution and free the webcam resource. Is there a way to detect when HoloLensARToolkit stops?

ericalbers, 2017-08-12:
the pause looks like it ends up calling StopFrameReaderAsyncTask, in ARUWPVideo.cs which will change
controller.status to ARUWP.ARUWP_STATUS_CTRL_INITIALIZED;  when its done

qian256, 2017-08-14:
@Baxab The pause is a .NET asynchronous function, which returns before it is really done. As mentioned by @ericalbers , you can check controller status because `controller.status = ARUWP.ARUWP_STATUS_CTRL_INITIALIZED;` is called after the async function really done.

closed 2019-01-11 by qian256