qian256/HoloLensARToolKit #25: latestTrackingInfo.trans question

ericalbers, 2017-08-12:
Hello!  Thanks again for all your work on this,
I'm trying to understand the 4x3 matrics which is returned by aruwpUpdate,
aruwpUpdate is passed a ARUint8 *frame, and ends up creating a matrics from the frame in 
latestTrackingInfo.trans in Update....
What does the matrics represent?  Is it the screen coordinate of the icon in the frame?  Then that gets translated to world coordinates?
The matrics is 4x3 and I don't see a lot of zeros, so it must have some meaning other than the screen points where the image was seen?
Thanks!

qian256, 2017-08-14:
The 4x3 matrix is 3x3 rotation and 3x1 translation. It represents the pose of the marker with respect to the camera coordinate system. It is not screen coordinates (x,y). It has rotation and depth, because the shape of marker and the size of the marker are predefined.

qian256, 2017-08-14:
You may find this an interesting reading: http://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html#solvepnp

Essentially, ARToolKit is doing similar thing to OpenCV solvepnp function.