qian256/HoloLensARToolKit #39: Slight judder in tracked object when moving head

DanAndersen, 2019-01-28:
Hi,
I noticed that there is a small one-frame judder that happens when the user's head moves. It's noticeable when a marker is kept in the same location and the head rotates back and forth (and it does not appear when the head is kept still and the marker is moved back and forth).

The cause seems to be the OnFrameArrived() function in ARUWPVideo.cs, where the line:
```
Interlocked.Exchange(ref _cameraToWorldMatrix, cameraToWorldMatrixAsFloat);
```
is being called too early. It leads to the case where the locatable-camera-to-marker matrix is being set one frame before the world-to-locatable-camera matrix is set, which causes the judder.

In my own tests I tried moving it to just before setting signalTrackingUpdated:
```
Interlocked.Exchange(ref _cameraToWorldMatrix, cameraToWorldMatrixAsFloat);
signalTrackingUpdated = true;
```
and the issue seemed to be resolved.

qian256, 2019-02-15:
Hi Dan, I noticed another strange behavior: when the head rotates left and right, the display of tracked object moves slightly back and forth (depth). Do you see the same issue?

DanAndersen, 2019-02-15:
Yes, I've noticed that issue as well, but I don't know what would be the cause. Let me know if you find anything out about it :/

qian256, 2019-02-15:
I have done some local fix. Since it seems to be a common issue now, I will clear my code and push it.
I think the main issue is that, the transform of dummy object, and the transform between marker and dummy object, update separately. The idea to fix is that, I parsed the cameraToWorldMatrix to ARUWPController to fire the processing of the frame, and then when the detection is done, the ARUWPController parsed the cameraToWorldMatrix to ARUWPMarker via the latestTrackingInfo (one additional field is added to this structure), then the ARUWPMarker updates its world transform directly. This seems to fix the issue.

DanAndersen, 2019-02-19:
Hi @qian256,
Is there a chance I could get a copy of your updated local code in a gist or something? I am planning on conducting a user study soon that involves this tracking as part of my system, and would love to be able to have the most stable tracking for that.

qian256, 2019-02-19:
@DanAndersen I will send you via email (to andersed@purdue.edu)

qian256, 2019-03-28:
The code is pushed. Still there are some judder. I imagine the next thing to do is to add "lerp" capability to tracked objects.

closed 2019-03-28 by qian256