qian256/HoloLensARToolKit #43: Cube is far from marker (probably copy of #17)

rschurade, 2019-03-18:
I'm running the HoloLenARToolKitSingle demo. I'd say it works fine so far, marker is recognized and it exactly follows any rotation or tilt of the marker. However it is roughly 20-30cm away from the marker. When I hold the marker in my hand and extend the arm it's about halfway between hand and eye.

Is it possible to adjust for that offset and can you advice where and how to do it?

doughtmw, 2019-03-18:
Have you performed camera calibration of the HoloLens RGB camera that you're using for tracking? 

If not, check out the code to calibrate the RGB camera [here](https://github.com/qian256/HoloLensCamCalib/tree/c953b85ebc21234c0f2b163e5f0d40257b7bae02).  You need to print off a checkerboard pattern, stick it on a planar board and take a bunch of pictures of it with the HoloLens camera (using the desired camera resolution which you will use for subsequent tracking). You can then use Long's code to estimate the intrinsic and distortion matrices, convert them into ARToolKit format and update them within the tracking code in this repo. Here is an example of camera calibration using OpenCV [link](https://docs.opencv.org/2.4.13.7/doc/tutorials/calib3d/camera_calibration/camera_calibration.html).

rschurade, 2019-03-18:
Thanks for the answer. I see this not really practicable as I need to put that into an app that is run on many  hololenses for any number of users. So I guess we'd need access to the calibration settings the user creates when creating the account on the hololens and then adjusting for that.

qian256, 2019-03-19:
@rschurade If the offset is constant, then you can manually offset all tracking results. But from your description, it is dependent on the distance of marker, than probably means that your HoloLens camera intrinsics are quite different from the default I provided (calibrating using my HoloLens).
You can calibrate the camera yourself as suggested by @doughtmw . In terms of updating camera matrix for each hololens. You can do something like this: http://longqian.me/2017/02/08/hololens-file-transfer/ 
The idea is to use Device Portal to upload files to hololens, and read from C# scripts. However the code may be obsolete due to API changes (I am not sure).

GuiGermano, 2019-09-20:
How can I take photos with a specific resolution of the camera?

actarus1975, 2021-12-06:
Hi @qian256, hi all, I still have this strange behaviour, that is the cube/virtual content stays far from the marker and tendentially moves towards the observer while he moves farther.
I calibrated the HoloLens 2 by using your applications/scripts and generated the .dat file, but even using the new matrix the results are very similar.
Can you suggest something to fix this issue?

actarus1975, 2021-12-07:
> I'm running the HoloLenARToolKitSingle demo. I'd say it works fine so far, marker is recognized and it exactly follows any rotation or tilt of the marker. However it is roughly 20-30cm away from the marker. When I hold the marker in my hand and extend the arm it's about halfway between hand and eye.
> 
> Is it possible to adjust for that offset and can you advice where and how to do it?
> 
> Edit: Hmm it's depending on the distance of the Hololens camera to the marker so there's probably no easy fix.

@rschurade I have the same identical issue. Did you find a way to fix it?