qian256/HoloLensARToolKit #44: Marker 0 no longer visible

SarahAmini, 2019-04-02:
Hi,
I just started using Hololens ARToolkit and was playing with the samples. I tried the sample for single marker and was looking at Hiro marker to see what happens...nothing happened. I turned my head several times, then was able to see a cube far in the distance, which was there for a second and then disappeared. I ran the code on VS 2017 to see what's happening, and realized the marker is not visible for more that 1 second, although neither me nor the marker move in the interval.
Am I doing something wrong? My unity version is 2018.3.11 and these are the messages I get in the console,back to back:

> (Filename: C:\buildslave\unity\build\Runtime/Export/Debug.bindings.h Line: 45)
> 
> 
> ARController (native): [info]Marker 0 no longer visible
>  
> (Filename: C:\buildslave\unity\build\Runtime/Export/Debug.bindings.h Line: 45)
> 
> 
> ARController (native): [info]Marker 0 now visible

yogihamdani, 2019-04-05:
Hi,
is the printed marker size the same as what it is stated in the Unity -> ARUWPController, look at the ARUWP Marker (script) -> Size in mm?

qian256, 2019-04-11:
From the messages, it looks like that the application is running fine. What is the configuration of ARUWPController in Unity?

SarahAmini, 2019-04-11:
> Hi,
> is the printed marker size the same as what it is stated in the Unity -> ARUWPController, look at the ARUWP Marker (script) -> Size in mm?

Yes, they were the same size. I even tested with my own marker that I made with the ARToolKit marker generator and printed in the same size.

> From the messages, it looks like that the application is running fine. What is the configuration of ARUWPController in Unity?

Sorry, I'm so new to this tooklit. Can you please guide me where I should  look for it?

qian256, 2019-04-14:
You can take a look at the wiki: http://longqian.me/2017/05/15/hololens-artoolkit-controller-v02/
It may be a little outdated, but should be able to give you some ideas.