qian256/HoloLensARToolKit #49: 我想请问你关于将这个插件适配到hololens2上的计划是怎样的？

zhuangce, 2019-12-26:
你好 ，我使用了您的代码，这在hololens1中表现的很好 ，但是当我尝试将其在hololens2中使用的时候，左上角的片元没有被着色，ar功能也没有开启 。随后我尝试将plugins中的ARTOOLKITUWP.all 改为
arm架构 ，不幸的是，这并没有什么用。我想请问你关于将这个插件适配到hololens2上的计划是怎样的，还有如果方便的话 ，我也想自己对其进行适配，但是我缺少相关的知识，你能告诉我一些modify的途径吗？如果可能的话。

qian256, 2020-01-13:
理论上需要重新编译ARTOOLKITUWP.all用arm64 然后可能相机参数和hololens1不一样，例如camera id
不过我现在没有hololens2，无法测试

LoisMonet, 2020-02-26:
Hi, do you succeed now with hololens 2 because this solution could be very useful for a project where I work. Else do you found others solution like maybe vuforia, visionlib or wikitude for hololens 2?

嗨，您现在是否成功使用hololens 2，因为此解决方案对于我工作的项目可能非常有用。 否则，您是否找到了其他解决方案，例如hololens 2的vuforia，visionlib或wikitude？

qian256, 2020-04-19:
Now it is supported with HoloLensARToolKit v0.3. Please check out

closed 2020-04-19 by qian256