qian256/HoloLensARToolKit #52: Demo doesn't work on Hololens2

whirwind, 2020-07-14:
Hi
I am finding ways to get video stream from HL2. I just loaded the demo for HL2 and deployed it to my device. There is no image on top left coner and Video/Track FPS display show "Infinity ms".

Unity asked me if I want to update Asset Database Version from 1.0 to 2.0. I tried both Version 1 and 2. In version 1 preview window return me a white screen and in version 2 there is nothing.

I am not sure why. Or what adjustment I need to do to make it work?

Unity: 2019.3.15f1
MRTK: 2.4 (I also tried to use original Unity project without MRTK)

ericalbers, 2020-07-15:
Also cannot get to work on my HL2, same issue, upper left camera preview is garbage/white, sometimes shows random black/white pixels, running on Unity 2019.4.3f1, can compile other HL2 programs and run fine.
Running HL2ARToolKitCube scene

ericalbers, 2020-07-15:
I'm seeing an error,
"ARController (native): [error]ARWrapper::ARToolKitVideoSource::open2(): error-failed to load camera parameters Data/StreamingAssets/HL2/1504x846.dat, calling close(), exiting returning false"

Here is no HL2 subdirectory under StreamingAssets....perhaps this is why the camera is failing

qian256, 2020-07-16:
@whirwind Please check the latest commits. The camera configuration files were not pushed to the Assets (ignored by .gitignore).

whirwind, 2020-07-16:
Thanks. The dome runs and I finnally see video stream of HL2 for the first time. I will try to make it work in my own project next.

closed 2020-07-16 by whirwind