qian256/HoloLensARToolKit #53: The tracking error is relevant to the distance between marker and divice.

MeganSage, 2020-07-22:
I have done the HoloLensCamCalib for better alignment . But actually the alignment error is not stationary , as the error gets bigger when I put marker away from the divice . Is there a optimal working distance for HoloLensARToolkit ？ Or Do I miss somthing else？

qian256, 2020-07-22:
Which device are you using, HL1 or HL2? Did you print the marker to be exact size as specified in the scene?

MeganSage, 2020-07-23:
> Which device are you using, HL1 or HL2? Did you print the marker to be exact size as specified in the scene?

Oopp! Never thought about the problem is the size of marker. I try the cube demo for HL2. I must have ignored the setup for the border size ! Thanks ! Bro!

MeganSage, 2020-07-23:
Thanks for your reply ! My device is HL 2, and I try the HL2ARToolKitSingle sample , the size of the marker I print  is 4cm (a 4x4 cm marker in A4 paper) . I guess the problem is not the size of the marker.  In your code,
The **position of the virtual object**  =  **TransMatrix  X  LocatableCameraToWorld   X  CalibrationMatirx**
I think it is the inaccuracy camera parameters I calibrate cause this magnifying error , as the error gets larger when the distance is increasing between HL2 and marker .  Maybe I need to calibrate my HL2 again for accuracy camera parameters.
By the way , I am a little confused about the Boader Size . As for the cube sample , the size of the cube I make is 6 cm , same as the virtual cube . And the size of the marker at the center of each face is 4 cm , so I have to change the Border Size parameter to  2/6 , right ?