qian256/HoloLensARToolKit #54: Not able to get tracking to work on HL1, VS2017

6peterlu, 2020-10-25:
Hello! Thanks for publishing this toolkit.

I am having issues getting it to track any marker (printed or on my computer screen). I have tried both the single (with both the kanji and hiro tags) and the multi.

I see no errors during build or deploy, and I can see the render fps stats. I can also see the rainbow cube, but it is just rendering in its static location just behind the starting camera coordinates. Do you have any tips on debugging this?

Thanks a bunch!

6peterlu, 2020-10-29:
Looking in the debug logs, I'm getting 
```
Unhandled 'Platform.COMException' exception caught! - 'The text associated with this error code could not be found.

Unable to load DLL 'ARToolKitUWP.dll': The specified module could not be found. (Exception from HRESULT: 0x8007007E)', Sender: '<null>'. Missing try/catch blocks.
 
(Filename: C:\buildslave\unity\build\PlatformDependent/MetroPlayer/MetroDebug.cpp Line: 46)
```
but I do have the .dll file in plugins/x86.

qian256, 2020-10-30:
Are you using .NET backend or IL2CPP?

6peterlu, 2020-10-30:
IL2CPP when I was using v0.3.

Ian-Sun, 2021-02-05:
Hi Peter. Glad to see your post. I'm having the same issue as well. Can I know if you manage to solve it or any potential trouble-shooting? Thanks!