qian256/HoloLensARToolKit #56: Any method to get RGB video stream in HL2?

whirwind, 2020-11-17:
Hello,

I noticed that in v0.3 feature grayscale is forced. I just need to get the RGB stream and handle every frame in Unity. it that possible?

Thanks

qian256, 2020-11-28:
For sure it is possible, you can look at the difference between v0.2 <-> v0.2 feature-grayscale, and apply back. It is mainly OnFrameArrived method and some settings of ARToolKit

zxzkf1992, 2021-05-13:
> For sure it is possible, you can look at the difference between v0.2 <-> v0.2 feature-grayscale, and apply back. It is mainly OnFrameArrived method and some settings of ARToolKit

Can you describe in detail how to obtain HL2 RGB stream in v0.3? I have tried to modify the source code of V0.3 (refer to the difference between v0.2 and v0.2 GrayScale). But still unable to get the RGB stream. And whenever I set the size of the framedata to four times the original size, an error occurs, causing the program to crash.

qian256, 2021-05-14:
The stream is always NV12, and in current implementation, we grab the grayscale part of NV12 image. To obtain RGB image, you need to conver the NV12 to RGB using UWP API SoftwareBitmap::Convert(), then you can access the pixel data.

zxzkf1992, 2021-05-14:
> The stream is always NV12, and in current implementation, we grab the grayscale part of NV12 image. To obtain RGB image, you need to conver the NV12 to RGB using UWP API SoftwareBitmap::Convert(), then you can access the pixel data.

Thank you so much for your quick reply. I have used this function before，

var softwareBitmap = SoftwareBitmap.Convert(frame.VideoMediaFrame.SoftwareBitmap, BitmapPixelFormat.Rgba8, BitmapAlphaMode.Ignore);

I still haven't got the RGB stream of HL2, but you have given me a certain direction and I will continue to try.

zxzkf1992, 2021-05-14:
> The stream is always NV12, and in current implementation, we grab the grayscale part of NV12 image. To obtain RGB image, you need to conver the NV12 to RGB using UWP API SoftwareBitmap::Convert(), then you can access the pixel data.

As you said, v0.3 only got the gray scale part of NV12. Does it mean that frameData only stores gray scale data? Because I did not see the relevant code to obtain the gray scale in the OnFrameArrived method.

qian256, 2021-05-14:
HL2 has no RGB stream, only NV12 stream. In order to get RGB image, you can convert it using SoftwareBitmap API

zxzkf1992, 2021-05-17:
> HL2 has no RGB stream, only NV12 stream. In order to get RGB image, you can convert it using SoftwareBitmap API

Thank you for your help, I have got the RGB image data of HL2.

zxzkf1992, 2021-05-21:
> HL2 has no RGB stream, only NV12 stream. In order to get RGB image, you can convert it using SoftwareBitmap API

hello, I found that v0.3 does not support the ARM (32-bit) platform. Is there an ARToolKitUWP.dll for this platform? Thank you !