qian256/HoloLensARToolKit #58: HL2: Is the marker tracking working from the play editor of Unity (in holographic remoting)?

9and3, 2021-02-09:
Hello and thanks for your work and for updating this great tool for tracking also for HL2!
I was wondering if it's possible to run the tracking from the play button mode (holographic remoting in HL2) without deploying the app on the device. I am asking this because almost all the other QR/ArUco tracking system in Unity for HL2 have to be deployed on the device to update and show the right pose of the marker.
Thanks for your answer!

qian256, 2021-02-23:
No this feature is not supported yet

u4e91, 2023-09-26:
hi, I have the same question and requirement with you, did you find some QR/ArUco tracking system can run in holographic remoting mode?thx! @9and3