qian256/HoloLensARToolKit #63: Unity 2020LTS does not build project

ergo333, 2021-05-22:
Hi @qian256, I'm quite new with HoloLens and Unity developing.
I am trying to create an application that uses some markers attached to a real robot. I necessarily need ROS integration with Unity 2020 LTS. 

I tried starting from the tutorial scene ARToolkitSingle but Unity2020 is not able to build the project. The Log console provides the following error: "Assets\ARToolKitUWP\Scripts\ARUWPVideo.cs(723,20): error CS0103: The name 'WorldManager' does not exist in the current context". 

I do not know how to fix this problem, has anyone solved it yet?

sLaderoute, 2021-06-03:
Hello! I was running into the same issue.
It seems UnityEngine.XR.WSA.WorldManager.GetNativeISpatialCoordinateSystemPtr() is no longer supported.

I got around this by commenting out the trouble line and replacing it with this:
`worldOrigin = SpatialLocator.GetDefault().CreateStationaryFrameOfReferenceAtCurrentLocation().CoordinateSystem;`
It directly sets the worldOrigin rather than dealing with an IntPtr at all.
[This is where I found the snippet that gives the SpacialCoordinateSystem](https://forum.unity.com/threads/unityengine-xr-wsa-worldmanager-is-obsolete-what-is-the-alternative.1062509/)

ergo333, 2021-06-13:
Oh thank you @sLaderoute, very nice. 
I will try this modification!

closed 2021-06-13 by ergo333

shoukailiang, 2022-01-08:
> @sLaderoute when I make this change I can build and deploy. But once the program is running on hololens- I lose the virtual cube when i look at the printed cube marker. Did you encounter this?

Hello, @soudehchegini .Did you solve it .

soudehchegini, 2022-01-09:
@sLaderoute thankyou for checkin in. This was a while ago now and Ive moved on from WorldLocking. However, I did run some WL programs so I must've found a way to make it work. I recall your suggestion above was very helpful. thankyou