qian256/HoloLensARToolKit #67: Has anyone succeeded in unity 2020?

shoukailiang, 2022-01-08:
Has anyone succeeded in unity 2020?

USTBVEO, 2022-04-11:
I tried to build on Hololens but it turned to the error : Assets\ARToolKitUWP\Scripts\ARUWPVideo.cs(723,20): error CS0103: The name 'WorldManager' does not exist in the current context
[image]
I searched and find that 'WorldManager' doesn't exist in Unity 2020.3.
[image]

petergu684, 2022-05-11:
In Unity 2020 (and all other versions using XRSDK), replace the line with following:
```cs
 WorldOriginPtr = UnityEngine.XR.WindowsMR.WindowsMREnvironment.OriginSpatialCoordinateSystem;
```
PS: It won't work in Unity 2021.2 or later where OpenXR is the only supported pipeline.

USTBVEO, 2022-10-18:
> In Unity 2020 (and all other versions using XRSDK), replace the line with following:
> 
> ```cs
>  WorldOriginPtr = UnityEngine.XR.WindowsMR.WindowsMREnvironment.OriginSpatialCoordinateSystem;
> ```
> 
> PS: It won't work in Unity 2021.2 or later where OpenXR is the only supported pipeline.

Thanks and I did that where I found that the hologram moved as I turned around my head. The hologram was not set to a fixed position. How can I tackle this problem? Thanks a lot!